richonguzman/LoRa_APRS_Tracker #219: Lilygo Lora32 ttgo-t-beam-v1_2 Power Off not possible with Power Button?

aendes, 2025-01-12:
Hello,

i have a flashed ttgo-t-beam-v1_2 to my Lilygo Lora32 as Tracker (latest firmware from git).
This module has PMU(?) AXP2101.

If it is running and i press the Power Button (long/short) nothing happens.
if i hit 3x times menu and Long Press it powers off.
If i then press the Power Button it powers back on.

As far as i know the other trackers shut off, wenn pushing the Power Button.
Is this a special behaviour of this Tracker/Device?

Thanks for your great work

Andreas

RadioMan57, 2025-01-12:
My T-Beam V1.0 takes about 6 seconds to shut it off with the power button.
Try again and be patient...

Markus HB9BRJ

richonguzman, 2025-01-12:
POWER BUTTON works for all T-Beams (maybe your buttons is bad?)

aendes, 2025-01-12:
Thanks @RadioMan57, i have to push 6 seconds...

So everything works like expected

closed 2025-01-12 by aendes