richonguzman/LoRa_APRS_Tracker #220: Heltec V2 error code -16

w5nmr, 2025-01-19:
My question:
Does the program work with Heltec V2? I get error code -16 when sending.
Thanks !

richonguzman, 2025-01-19:
hello

the tracker should work with Heltec V2, which version of firmware are you testing?
- DIY + GPS
- TNC

w5nmr, 2025-01-20:
Hello 
thanks for the quick response

[11:44:26][INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[11:44:26][INFO][Main] Version: 2025.01.11
[11:44:26][WARN][Main] GPS disabled
[11:44:26][INFO][LoRa] Initializing SX127X ...
[11:44:26][INFO][LoRa] LoRa init done!
[11:44:26][IN��%�Օѽ�ѡu�Bluetooth init done!
[11:44:27][INFO][Main] Setup Done!
[11:44:29][INFO][Bluetooth] Client connected !
[11:46:40][INFO][LoRa Tx] ---> DL1EEM-4>APDR16,WIDE1-1:=5315.00N/00811.54E4000/000/A=000242 APRSdroid LoRa
[11:46:40]failed, code -16

no transmission signal

richonguzman, 2025-01-20:
can you post the Lora configuration you used? changed anything of it?

the LoRa Module init correctly as the log shows.

does your board "listen" packets ?

w5nmr, 2025-01-20:
hello

it doesn't receive anything either

the same board works correctly as an i-gate
sends and receives

RADIOLIB_ERR_SPI_WRITE_FAILED   (-16)
 	Real value in SPI register does not match the expected one. This can be caused by faulty SPI wiring.

I don't understand. If there is no solution, please close and delete
vy 73 Peter

richonguzman, 2025-01-30:
i have not forgotten this.

the same board with the same pin definitions work as iGate. But not as Tracker TNC? right?

w5nmr, 2025-02-01:
Hello,
thank you very much for your effort!

Everything works as i-gate. send and receive via RF.
As a tracker the same board comes with the error .other software also works perfectly. It is not necessary to continue looking for the error. I use another board as a tracker. Thank you!
vy 73 Peter 
Google translation

richonguzman, 2025-02-01:
ok then I will close this

richonguzman, 2026-10-02:
solved in v4.0

closed 2026-10-02 by richonguzman