richonguzman/LoRa_APRS_Tracker #221: Request configuration: add box to set BT name

ik5xmk, 2025-01-19:
Hi Ricardo, if there are two trackers available, both with the same call but different ssid, eg. callsign-7 and callsign-9, via BT they are always seen as callsign-BT and to manage and recognize them you need to unpair them and associate them again, turning off the one you don't want first.

Is it possible to have, in the WebGUI, under the BT selection also the possibility to specify a name for the tracker? Callsign-BT is fine by default but if I have two devices maybe I prefer to call them dev001-BT and dev002-BT to know immediately which one I am associated with.

Also because, otherwise, I only notice the error because the tracker does not communicate with the smartphone.
I am thinking of applications like APRSDroid.

Thanks, 73. David

richonguzman, 2025-01-20:
this is a nice fix/idea to implement

the problem I have is that when you change between callings in tracker (with user button) I would need to reset the BT/BLE connection to reflect the change unless I change the way BT/BLE is saved in the web to enable any name you would like to use beside the callsign

ik5xmk, 2025-01-20:
Or, simply, if you use multiple callsigns the BT connection always remains the same (it makes sense, I change the call), but if instead I change the tracker the need to change the callsign is no longer needed, so you can disable the possibility of having multiple callsigns if you customize the name of the BT connection. 

my 2 cents for having flexibility. Thanks Ricardo you see for yourself of course, but at the moment I think it is very effective to have multiple associated trackers, for external device management.

richonguzman, 2025-01-20:
so having just the first callsign in memory would be enough?

ik5xmk, 2025-01-20:
For my way of operating yes, but I don't want to go against what has already been implemented and probably in use by someone.

Precisely for this reason I was thinking of a situation like this:

- **I activate (switch on)** the change of the bluetooth name, I inhibit the possibility of having 3 callsigns and I can edit/insert the name of the device as I want. I will use only a single callsign but I will see, for example, dev001-BT as the name of this tracker.

- **I deactivate (swith off)** the change of the bluetooth name, it remains as the name of the device "callsign-BT" and I can use, everything as it is now, the 3 callsigns and use the hardware button.

So with a switch I have both possibilities.

richonguzman, 2025-01-30:
Next update (just a few days) will have this enabled as BT control

- BT active
- BT Device Name
- BLE active /or BT classic
- KISS active / or use TNC2

closed 2025-01-30 by richonguzman