richonguzman/LoRa_APRS_Tracker #223: weather data as telemetry for private mesurement

iw2ejh, 2025-01-22:
We had the WX funtion added to the LoRa_APRS_Tracker, this is nice, but in some case, the temperature and humidity collected are not related to the local external Wheater but are related to a dedicated environment, like Car, electric box, dedicated room, etc. Would it be possible to add a switch in the configuration in order to be able to select if we want to send these data as WX data or as Telemetry? Using telemetry format will keep track of the value, and keep them private, (not mixing up with WX data from wheather stations) I think this may help in keeping WX data clean from wrong/indoor values.

richonguzman, 2025-01-22:
this could be a good mod to the tracker firmware to avoid "faulty" Wx data... will check about it

Jan-OK2, 2025-01-27:
That may be similar like my [https://github.com/richonguzman/LoRa_APRS_iGate/issues/241] : sending WX data (especially temperature in C degrees) as telemetry, APRS WX format or both. And may be implemented in iGate's FW also. There's no problem with sending more pakets directly into APRS-IS.

richonguzman, 2025-12-31:
> hi, how do you connect the aht20+bmp280 module to the lilygo t-beam having already connected oled on pins 21 and 22??

you connect it in parallel: I2C connections has internal addresses and can be connected without any problem in parallel

iz7exi, 2025-12-31:
Sorry for all the questions. Can the tracker be accessed via Wi-Fi once programmed, or do I have to download an app?
I noticed there are no settings for my home Wi-Fi connection.