richonguzman/LoRa_APRS_Tracker #225: Weather report is not working on 11 January firmware

iuliandj, 2025-01-22:
Hello,

I discover that on 11 january tracker firmware, Weather report is not working. You can sent query but with no response. 
In firmware version from 21 December I receive response from CA2RXU-15.

Thank you!

richonguzman, 2025-01-22:
Hi, which tracker callsign-said is this?

does it have a gps beacon listened by a igate?

igate must hear the beacon to then send the answer back to it

iuliandj, 2025-01-22:
tracker callsign is YO7MGH
Yes, his position was heard by an Igate.
With firmware from 21 december works great. Now is installed this version on tracker.

richonguzman, 2025-01-22:
https://aprs.fi/?c=message&call=YO7MGH 

all packets have been processed and should be sended back at you without any problem.

version of 21 Dec vs 11 Jan have no difference on the packet reception.

can you check if YO7MGH-6 sent the packet from APRS-IS to your tracker?

iuliandj, 2025-01-22:
I checked now on aprs.ri but I can not find that information.
What I can tell you is that from test with firmware version 11 January, I did not receive response from your WX station by YO7MGH-6.
On your link above, I can see a query from 18.01. In that day I had 11 January firmware.  Today at 14:30 I thnk I also had 11 Jan vers and after that, at 15:00,  I change it to 21 Dec vers. and I started to receive answer from WX by IG. 
Your WX call appears at Last RX now. With 11 Jan vers. it does not, so query was not sent from IG to RF, I guess.

richonguzman, 2025-01-22:
I guess you should go back to the newest firmware and test again... as the firmware has not change at all the way it receives packets

please test and let me know

iuliandj, 2025-01-22:
Do you recommend to erase and use web installer or update by WiFI Config AP by uploading the file ?

richonguzman, 2025-01-22:
Tracker can be updated only by webflasher

but you can save the current configuration on the WebUI and download the configuration.json file which could be uploaded to restore fast the new tracker

iuliandj, 2025-01-22:
Ok. That is what I did today.
Thank you!

iuliandj, 2025-01-24:
I confirm that it works with 11 Jan vers on 2 test from 3. 
 The 22 Jan vers. is not on web installer so not testet.

closed 2025-02-19 by richonguzman