richonguzman/LoRa_APRS_Tracker #226: Battery percentage stuck issue

iuliandj, 2025-01-22:
Hello,
I notice in firmware from 11.01.2025, on display it remains stuck 91% battery percentage.
The battery voltage value is modifying but percentage is not. This happened after a complete discharge of battery and charge it to 91%, after that I decoupled from charger because of need.
Even if I complete charge the battery it appears 91%.
Today I installed 21 december firmware and 91% was still on display even if memory was erased :-)
I get rid of 91% by pulling out the battery, powered the device by usb, and push back the battery inside. Everything went back to normal but let's see for how long. :-)
I did not opened the tracker case before because is a little bit hard to open and I do not what to damage the board.

Thank you verry much!

richonguzman, 2025-01-22:
hi,

which boards is this?

iuliandj, 2025-01-22:
> hi,
> 
> which boards is this?

T-beam V1.2

richonguzman, 2025-01-22:
T-beam V1.2 uses the AXP2010 power management chip

this chip as a bug that sometimes needs a reset:
you need to take out the usb cable connector and the battery so that it has no power at all.

then you can power it with battery or usb and all battery voltages will be good.

please test it and confirm

iuliandj, 2025-01-22:
I tested today and I conform that after I pulled out the battery, and push back in, I get rid of 91% stuk.

closed 2025-01-22 by richonguzman