richonguzman/LoRa_APRS_Tracker #228: Battery not recognized by the board (Rev1.1 20210222)

oe9fwv, 2025-02-03:
I have successfully flashed the firmware to my TTGo [Rev1.1 20210222] but I am not sure if I chose the correct board when flashing it.  My problem is that on the screen it says "no battery conected" with a 18650 battery connected to the board. When I connect the board to USB it says Bat 0.29V (charging). 
I programmed the board as ttgo-t-beam-v1. I am not sure if this is the correct setting for my board. 

thank you Ricardo and contributors for the great work!

73! Werner oe9fwv

richonguzman, 2025-02-03:
Hi Werner,

if your board is the T-beam V1.2 you should take out USB connection and also Battery from the board.

this will reset the AXP2101 powermanagment chip and when putting the battery again into the T-beam it will start and recognize it as it should

let me know if this works

oe9fwv, 2025-02-03:
hi Ricardo, 
thank you very much for the quick answer!
I re-flashed the board, this time with ttgo-t-beam-v1_2. Removed USB, removed battery. Re-inserted the battery (18650 on the backside of the board) and on the display I still read "no battery connected".
When I insert the USB plug, it says "Battery 0.29V 0%"

73! Werner oe9fwv

richonguzman, 2025-02-03:
Is your board the Tbeam v1 or v1.1 or v1.2?

oe9fwv, 2025-02-03:
I do not know for sure. On the board it says 
T22_V1.1
on the backside LILYGO
How can I find it out?

73! Werner

richonguzman, 2025-02-03:
So you should flash the Tbeam v1 firmware

oe9fwv, 2025-02-03:
Ricardo, 
thank you very much, this solved the problem! 
It shows the correct voltage and charging status.

73! Werner oe9fwv

richonguzman, 2025-02-03:
Great news

closed 2025-02-06 by oe9fwv