richonguzman/LoRa_APRS_Tracker #229: Weather station symbol on OLED display

oe9fwv, 2025-02-08:
hi Ricardo, 
I added a BME 280 module to my tracker and set the APRS symbol to /_ - Weather station.  (oe9fwv-11)
Works great on aprs.fi. Would it be possible to add the WX symbol to the OLED display like bicycle and car?

I have another question related to the weather station: 
The raw data in aprs.fi show this error: 
2025-02-08 06:20:38 CET: [OE9FWV-11](https://aprs.fi/?c=raw&limit=&call=OE9FWV-11)>APLRT1,[HB9BB-10*](https://aprs.fi/?c=raw&limit=&call=HB9BB-10),qAO,[OE9MNR-11](https://aprs.fi/?c=raw&limit=&call=OE9MNR-11):!/6WtcPtip_H{Q.../...g...t070h26b10228 SNR=-7dB RSSI=-84db [Packet corrupted by iGate]

Is this a problem of the Igate used at HB9BB?
My Igate with your firmware loaded does not show an error:
2025-02-08 06:20:32 CET: [OE9FWV-11](https://aprs.fi/?c=raw&limit=&call=OE9FWV-11)>APLRT1,WIDE1-1,qAO,[OE9FWV-L4](https://aprs.fi/?c=raw&limit=&call=OE9FWV-L4):!/6WtcPtip_H{Q.../...g...t070h26b10228

73! Werner oe9fwv

richonguzman, 2025-02-08:
Hi Werner,

as a quick review:
OE9MNR-11 or HB9BB-10 are modifying the received packet but I could not find a way to know for sure which one of them did it.

so yes, your tracker has rejected packets when entering through this path because of this packet modification.

OE9FWV-L4 is working fine ;)  (although I would recommend to use AX-25 valid SSIDs for it to have full RF Tx possibilities : SSID from 0-15 instead of L4)

as for the symbol, which one should I "draw" ? try to copy the WX blue circled symbol somehow?

oe9fwv, 2025-02-08:
hi Ricardo, 
as for the symbol: Blue circle "WX" would be nice :-) 

I know the sysops of oe9mnr-11 and hb9bb-10 - maybe I can convince them to update to your igate software. 
Regarding the Lora Igate Extension I have seen that L4 is popular but indeed not covered in the APRS specs. I should probably change it to -10. thank you for the hint!

73! Werner oe9fwv

Mane76, 2025-02-08:
> as a quick review: OE9MNR-11 or HB9BB-10 are modifying the received packet but I could not find a way to know for sure which one of them did it.
> 

Hi Ricardo, 
OE9MNR-11 works as iGate, and HB9BB-10 work as pure digi, but show the iGate symbol. 

I am absolutely sure that OE9MNR-11 is modifiying the packets by examining the raw data. 
All packets purely received from OE9MNR-11 have the modification with SNR/RSSI.

HB9BB-10 does not modify anything but should be configured to identify as Digi to appear correctly on the map.

BR
Mane

richonguzman, 2025-02-19:
wx symbol has been added, please check it on last Version

oe9fwv, 2025-02-20:
Thank you! Installed it, looks nicht!

closed 2025-02-20 by oe9fwv