richonguzman/LoRa_APRS_Tracker #230: Can somebody help me with build in VCS

Jan-OK2, 2025-02-12:
I am not good familiar with using (and programming in) VCS and now my attemps to build failed with this error:

c:/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot open map file C:/********************/LoRa_APRS_Tracker-main/.pio/build/heltec_wireless_tracker/firmware.map: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\heltec_wireless_tracker\firmware.elf] Error 1 

That does do last two versions of LoRa_APRS_Tracker. With the olders one I had no problem. And the same problem is now with compiling for another board too. Web flashing is OK.
Any idea?

richonguzman, 2025-02-12:
you have to share way more info to really understand what you are doing there...

Jan-OK2, 2025-02-13:
Okay, I'll try. I've been using VCS for editing, building and uploading various projects as a "user", with little experimentation with code and configuration.  After the necessary editing, I selected the Platformio icon, selected the board - General/Build/Build Filesystem-
/Upload. Now I downloaded the .zip of the new version 2025.02.09 and completely unzipped it to the original directory (rewrited all older). I chose the same procedure as I described and the result was this. I tried selecting other boards, and always the same result.

[image]

richonguzman, 2025-02-13:
this looks like you added/changed/modded some files or even delete some files in there...

save the mods on the files you did and download the new updated code and paste your mods into it to validate it all runs.

anyway: this seems more like a you+VSCODE manipulation error

Jan-OK2, 2025-02-13:
I deleted complete directory of this project (with my mods) , downloaded the new, without mods, placed it back and have open it in VCS and tryied to build... Tryied open as new project too. Always the same error. Zasláno z Outlooku pro AndroidOd: Ricardo Guzman (Richonguzman) ***@***.***>Odesláno: čtvrtek, února 13, 2025 12:39:07 odp.Komu: richonguzman/LoRa_APRS_Tracker ***@***.***>Kopie: Jan-OK2 ***@***.***>; Author ***@***.***>Předmět: Re: [richonguzman/LoRa_APRS_Tracker] Build error: "/firmware.map: No
 such file or directory" (Issue #230)
this looks like you added/changed/modded some files or even delete some files in there...
save the mods on the files you did and download the new updated code and paste your mods into it to validate it all runs.
anyway: this seems more like a you+VSCODE manipulation error

Jan-OK2, 2025-02-13:
Yes, I did it - in Platformio selected "Import new Arduino project". I selected directory and board (tryied T-Beam)...

Jan-OK2, 2025-02-13:
Yes, it seems so. 
I just don't understand that there are very many boards in the roll menu, but the TTGO T-Beam with no version designation and the board I want to test, the Heltec Wireless Tracker, is not there at all. Then in the project menu itself these boards can be individually selected (V1.1, V1.2., heltec wireless tracker...)- even with their options (build, upload....)

richonguzman, 2025-02-14:
it seems you are not importing it fully or the right way... did you see this: ???

https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/01.-Installation-Guide-%23-Guia-de-Instalacion

Jan-OK2, 2025-02-15:
Yes, many times. Now I can't even compile your igate project (same error), while other projects (Cubecell tracker by DJ1AN) work normally. I don't know what to try anymore - I completely reinstalled VCS, Platformio, re-imported your project many times, tried to build for another board, but the build always ends up the same. Thanks for your patience!

Jan-OK2, 2025-02-19:
Still no progress! Is the latest Build working normally for all of you?
I tried the usual procedure - "Import arduino project - select board TTGO LoRa32-OLED v2.1.6 - select project directory" and then with or without configuration changes tried building for a specific board. All projects work this way: e.g. from DJ1AN, OE5BPA, but also WX Station of CA2RXU. I can't compile these Igate and Tracker only.

Jan-OK2, 2025-02-19:
Solved!
There was a directory with a Czech name in the path to the project directory. For Plarformio there must be only ASCII characters in the path!

richonguzman, 2025-02-19:
got it, nice info!!!

and great that you finally solved it ;)

closed 2025-02-19 by richonguzman