richonguzman/LoRa_APRS_Tracker #231: update V2.1.4 -> V2.2 Turn off function from menu is broken

manitwo10, 2025-02-13:
I'm using standard Heltec Tracker Board.
After update from 2.1.4 to 2.2 the power off from menu is broken. After turn off the tracker by menu (3 short | 1 long press) an reset will immediately start the tracker again. No way to start deep sleep anymore.

manitwo10, 2025-02-13:
I did upgrade with platformio, no change in config nor use any web interface. When upload back to former version deep sleep works again as expected.

richonguzman, 2025-02-13:
nothing changed on the POWER_Utils::shutdown() process....

@9M2PJU @manitwo10  are you powering it off (deep sleep) from the "3 button push" menu?

lorusLD, 2025-02-13:
Same problem.

What I did first back up all config, did clean install of V2.2 with erase over web installer.

After restar uploaded  config and deep sleep not working.

9M2PJU, 2025-02-13:
> nothing changed on the POWER_Utils::shutdown() process....
> 
> [@9M2PJU](https://github.com/9M2PJU) [@manitwo10](https://github.com/manitwo10) are you powering it off (deep sleep) from the "3 button push" menu?

Yes. I think something is broken because power_utils not changed.

richonguzman, 2025-02-13:
this is really strange:

i flashed mine, changed Callsign and BT data , saved.... then I pressed 3 times... forced deep sleep and it sleeps

[image]

manitwo10, 2025-02-13:
> I'm using standard Heltec Tracker Board.

it seems not the case on all devices. When upgrading TTGO T-Beam V1 it works as expected.
Heltec Tracker only has 2 buttons, T-Beam 3 ... not sure if this make a different.

richonguzman, 2025-02-13:
can you @manitwo10 @9M2PJU @lorusLD test just flashing, changing only CALLSIGN, save and use it??

then try make it sleep? (this worked for me "as usual ")

lorusLD, 2025-02-13:
Problem is on Heltec Wirelles tracker

Lilygo T-Beam no problem

manitwo10, 2025-02-13:
Okay, I did fresh install by WebFlasher incl. erase all, then set my call by webui and restart. Then set menu to deep sleep but still device restarts again.

This is the logfile from the console
```
--- Terminal on COM42 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[   890][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not Station
[INFO][Main] Version: 2025.02.12
[INFO][LoRa] Initializing SX126X ...
[INFO][LoRa] LoRa init done!
[INFO][Main] Setup Done!
[INFO][LoRa Tx] ---> DL4MN-3>APLRT1,WIDE1-1:!/4;^BQ5@p[!!Q
[WARN][Main] SHUTDOWN !!!

assert failed: npl_freertos_mutex_pend npl_os_freertos.c:1580 (mu->handle)

Backtrace: 0x403782d6:0x3fceb9a0 0x40381601:0x3fceb9c0 0x40387e71:0x3fceb9e0 0x4202ec39:0x3fcebb10 0x4202836c:0x3fcebb30 0x42028383:0x3fcebb50 0x4202a09f:0x3fcebb70 0x4202df62:0x3fcebb90 0x42020dd7:0x3fcebbb0 0x420035a9:0x3fcebbd0 0x420187b3:0x3fcebbf0 0x42009799:0x3fcebc30 0x42003d70:0x3fcebcd0 0x42003dfe:0x3fcebcf0 0x42002cc2:0x3fcebd10 0x4204e3bd:0x3fcebd70
  #0  0x403782d6 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x40381601 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x40387e71 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x4202ec39 in npl_freertos_eventq_get at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/porting/npl/freertos/src/npl_os_freertos.c:1430
  #4  0x4202836c in ble_hs_wakeup_tx_conn at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:274 (discriminator 1)
  #5  0x42028383 in ble_npl_eventq_put at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:514
      (inlined by) ble_hs_sched_start_stage2 at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:514
      (inlined by) ble_hs_event_start_stage1 at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:582
  #6  0x4202a09f in ble_hs_startup_reset_tx at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_startup.c:301
      (inlined by) ble_hs_startup_go at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_startup.c:311
  #7  0x4202df62 in ble_store_config_persist_our_secs at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/store/config/src/ble_store_nvs.c:594 (discriminator 15)
  #8  0x42020dd7 in NimBLEDevice::deleteClient(NimBLEClient*) at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/NimBLEDevice.cpp:264
  #9  0x420035a9 in BLE_Utils::sendToLoRa() at src/ble_utils.cpp:137
  #10 0x420187b3 in POWER_Utils::externalPinSetup() at src/power_utils.cpp:277 (discriminator 1)    
  #11 0x42009799 in KEYBOARD_Utils::rightArrow() at src/keyboard_utils.cpp:462
  #12 0x42003d70 in BUTTON_Utils::isrSinglePress() at src/button_utils.cpp:52
  #13 0x42003dfe in BUTTON_Utils::doublePress() at src/button_utils.cpp:41
      (inlined by) BUTTON_Utils::doublePress() at src/button_utils.cpp:34
  #14 0x42002cc2 in String::len() const at C:/Users/Marco/.platformio/packages/framework-arduinoespressif32@3.20016.0/cores/esp32/WString.h:322
      (inlined by) String::substring(unsigned int) const at C:/Users/Marco/.platformio/packages/framework-arduinoespressif32@3.20016.0/cores/esp32/WString.h:264
      (inlined by) loop() at src/LoRa_APRS_Tracker.cpp:204
  #15 0x4204e3bd in uartDetectBaudrate at C:/Users/Marco/.platformio/packages/framework-arduinoespressif32@3.20016.0/cores/esp32/esp32-hal-uart.c:949

ELF file SHA256: df7bdb71c0144681

Rebooting...
ESP-ROM:esp32s3-20210327
```

richonguzman, 2025-02-13:
did you have the board paired with BLE to iPhone app when trying to do the "shut down" process?

> Okay, I did fresh install by WebFlasher incl. erase all, then set my call by webui and restart. Then set menu to deep sleep but still device restarts again.
>
> This is the logfile from the console [...]

did you have the board paired with BLE to iPhone app when trying to do the "shut down" process?

manitwo10, 2025-02-13:
... no, default configuration and only call added.
```
"bluetooth": {
		"active": false,
```

richonguzman, 2025-02-13:
> ... no, default configuration and only call added.
> 
> ```
> "bluetooth": {
> 		"active": false,
> ```

besides False, for BT, any other configuration was being changed?

could it be that BLE is getting somehow an error?

manitwo10, 2025-02-13:
> besides False, for BT, any other configuration was being changed?

no, only all 3 callsigns were changed

richonguzman, 2025-02-13:
found the bug:

with the same settings you have... could you add this:  Config.bluetooth.active

like this in ble_utils.cpp line 477 ??

[image]

richonguzman, 2025-02-13:
V2.2.1 is out to test this fix ;)

lorusLD, 2025-02-13:
Not working for me...

Still same error...

richonguzman, 2025-02-13:
> Not working for me...
> 
> Still same error...

you got to give me more info please...  which version did you try? what did you input on webConfig ?

lorusLD, 2025-02-13:
Web installer -> then upload config file 

[TrackerConfigurationBackup_H_WT_S50V.json](https://github.com/user-attachments/files/18789758/TrackerConfigurationBackup_H_WT_S50V.json)

richonguzman, 2025-02-13:
> Web installer -> then upload config file
> 
> Used V2.2.1 version
> 
> [TrackerConfigurationBackup_H_WT_S50V.json](https://github.com/user-attachments/files/18789758/TrackerConfigurationBackup_H_WT_S50V.json)

I explained the info should be written from zero into webUI to test this, old config file wont work as BT part has changed

lorusLD, 2025-02-13:
Ok!

lorusLD, 2025-02-13:
I used Web Installer -> Heltec Wireless Tracker -> Version V2.2.1 -> Full erase and new install -> Web config ONLY call sign was entered and nothing else was changed

Still same error with Deep Sleep

This is fresh config from fresh clean install
[TrackerConfigurationBackup_V221.json](https://github.com/user-attachments/files/18790014/TrackerConfigurationBackup_V221.json)

manitwo10, 2025-02-13:
> I used Web Installer -> Heltec Wireless Tracker -> Version V2.2.1 -> Full erase and new install -> Web config ONLY call sign was entered and nothing else was changed
> 
> Still same error with Deep Sleep

I can confirm this, still the same restart after deep sleep on Heltec Tracker with full fresh Flash with V2.2.1 and minimal webUi input and restart.

richonguzman, 2025-02-13:
Can you post the serial output to see the problem?

manitwo10, 2025-02-14:
its from platformio console with V2.2.1 on Heltec Tracker 

```
--- Terminal on COM17 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[   897][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not [INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[INFO][Main] Version: 2025.02.12
[INFO][LoRa] Initializing SX126X ...
[INFO][LoRa] LoRa init done!
[INFO][Main] Setup Done!    
[WARN][Main] SHUTDOWN !!!

assert failed: npl_freertos_mutex_pend npl_os_freertos.c:1580 (mu->handle)

Backtrace: 0x403782d6:0x3fceba30 0x40381601:0x3fceba50 0x40387e71:0x3fceba70 0x4202ec39:0x3fcebba0 0x4202836c:0x3fcebbc0 0x42028383:0x3fcebbe0 0x4202a09f:0x3fcebc00 0x4202df62:0x3fcebc20 0x42020dd7:0x3fcebc40 0x420035a9:0x3fcebc60 0x420187b3:0x3fcebc80 0x42009799:0x3fcebcc0 0x42003d70:0x3fcebd60 0x42003dfe:0x3fcebd80 0x42002cc2:0x3fcebda0 0x4204e3bd:0x3fcebe00
  #0  0x403782d6 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x40381601 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x40387e71 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x4202ec39 in npl_freertos_eventq_get at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/porting/npl/freertos/src/npl_os_freertos.c:1430
  #4  0x4202836c in ble_hs_wakeup_tx_conn at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:274 (discriminator 1)
  #5  0x42028383 in ble_npl_eventq_put at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:514
      (inlined by) ble_hs_sched_start_stage2 at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:514
      (inlined by) ble_hs_event_start_stage1 at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:582
  #6  0x4202a09f in ble_hs_startup_reset_tx at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_startup.c:301
      (inlined by) ble_hs_startup_go at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_startup.c:311
  #7  0x4202df62 in ble_store_config_persist_our_secs at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/nimble/nimble/host/store/config/src/ble_store_nvs.c:594 (discriminator 15)
  #8  0x42020dd7 in NimBLEDevice::deleteClient(NimBLEClient*) at .pio/libdeps/heltec_wireless_tracker/NimBLE-Arduino/src/NimBLEDevice.cpp:264
  #9  0x420035a9 in BLE_Utils::sendToLoRa() at src/ble_utils.cpp:137
  #10 0x420187b3 in POWER_Utils::externalPinSetup() at src/power_utils.cpp:277 (discriminator 1)
  #11 0x42009799 in KEYBOARD_Utils::rightArrow() at src/keyboard_utils.cpp:462
  #12 0x42003d70 in BUTTON_Utils::isrSinglePress() at src/button_utils.cpp:52
  #13 0x42003dfe in BUTTON_Utils::doublePress() at src/button_utils.cpp:41
      (inlined by) BUTTON_Utils::doublePress() at src/button_utils.cpp:34
  #14 0x42002cc2 in String::len() const at C:/Users/zzzzzkbx/.platformio/packages/framework-arduinoespressif32@3.20016.0/cores/esp32/WString.h:322
      (inlined by) String::substring(unsigned int) const at C:/Users/zzzzzkbx/.platformio/packages/framework-arduinoespressif32@3.20016.0/cores/esp32/WString.h:264
      (inlined by) loop() at src/LoRa_APRS_Tracker.cpp:204
  #15 0x4204e3bd in uartDetectBaudrate at C:/Users/zzzzzkbx/.platformio/packages/framework-arduinoespressif32@3.20016.0/cores/esp32/esp32-hal-uart.c:949

ELF file SHA256: df7bdb71c0144681

Rebooting...
ESP-ROM:esp32s3-20210327
```

richonguzman, 2025-02-14:
> Backtrace: 0x403782d6:0x3fceba30 0x40381601:0x3fceba50 0x40387e71:0x3fceba70 0x4202ec39:0x3fcebba0 0x4202836c:0x3fcebbc0 0x42028383:0x3fcebbe0 0x4202a09f:0x3fcebc00 0x4202df62:0x3fcebc20 [...]

please confirm to test the same here:
only 3 callsign change and saved? then with this basic settings it won't deep sleep?

richonguzman, 2025-02-17:
@manitwo10 @9M2PJU @lorusLD please test V2.2.2 as it seems to fix all the BLE crashing problems which was generated when I changed a few things on how this is controlled.

I've tested it locally and it seems to work just fine

manitwo10, 2025-02-17:
I did upload V2.2.2 by platformio and the Heltec Tracker stays in deep sleep now, no reset.
For me this issue seems solved now.

lorusLD, 2025-02-17:
Hi!

I can confirm, all is now working as before! Case closed.

Thank you Ricardo!

Regards,
Lovro

richonguzman, 2025-02-19:
as this has been solved now I'm closing it ;)

closed 2025-02-19 by richonguzman