richonguzman/LoRa_APRS_Tracker #233: Woad to loraaprstracker

edegraaff, 2025-02-14:
Hi, does someone a working integration with woad? 
Approach:
Aprstracker settings, bluetooth enable, classic, enable use kiss. 
Woad : protocol packet, outgooing, KISS > bluetooh to the aprstrackr
Start connection: 
Woad shows green > and i see in the igate messages like these PD1EG>PD1EG: 
It seems i am not able to select an RMS channel i already reported this to Woad 
It seems it is working partly.

If i use aprs droit i am able to sent tekst messages and that is working
So what could be the issue not to be able to use woad? 
16:03:52	PD1EG-5>APDR16,WIDE1-1::PD1EG :test{1	-50	6.75
16:04:23	PD1EG-5>APDR16,WIDE1-1::PD1EG :test{1	-51	6.25
16:05:23	PD1EG-5>APDR16,WIDE1-1::PD1EG :test{1	-46	6
16:05:50	PD1EG-5>APDR16,WIDE1-1::PD1EG :ok{2	-48	6.75
16:06:20	PD1EG-5>APDR16,WIDE1-1::PD1EG :ok{2	-49	5.5
16:06:59	PD1EG-7>APLRT1,WIDE1-1:!/4.8TOY?&[ "G	-48	5.25

richonguzman, 2025-02-14:
i have been trying to get info everywhere to make WOAD/RADIOMAIL apps work with LoRa packets.

but the protocol, the way packets flow between station to gateway is not clear to me and I keep on looking

do you know something about this?

edegraaff, 2025-02-14:
i hope woad comes  with something, in theory it should work

richonguzman, 2025-02-14:
> i hope woad comes with something, in theory it should work

but have you contacted them?

edegraaff, 2025-02-14:
> > i hope woad comes with something, in theory it should work
> 
> but have you contacted them?

Yes i did ... :-) lets see what it brings.

richonguzman, 2025-03-18:
I think I will see "the light" soon with this but on another type of repository.

so this will be closed and news will be shared upon good news

closed 2025-03-18 by richonguzman