richonguzman/LoRa_APRS_Tracker #235: Problem APX2101 init failed!

rn1m, 2025-02-15:
Hello everyone!

The tracker worked fine all this time, but at some point I noticed that it stopped loading. I reflashed it with a newer firmware, both via the web and via the platformio, rolled back to an older firmware, the error is as follows:

What else can I do? Is this a hardware error, did something burn out?

73, RN1M

[image]

S57PNX, 2025-02-15:
this looks like you are loading the incorrect firmware platform for your hardware. Please double check that you are choosing the correct hardware. 

If it still doesn't work, we'll need more details to help you (a photo of your hardware and the platform that you chose).

rn1m, 2025-03-02:
Hello.

My board LILYGO T-beam AXP2101-V1.2

[image]

WebFLASH config:

[image]

Erase

[image]

[image]

Error:

[image]

[image]

Ideas?

richonguzman, 2025-03-02:
@rn1m please take out USB connection and battery from the T-beam board. Then connect the Battery again and the USB. It should work

Explanation: AXP2101 power management chip of the T-beam v1.2 sometimes needs a reset process to work after a firmware flashed process.

rn1m, 2025-03-08:
[image]

Thanks for help! 73!

closed 2025-03-08 by richonguzman