richonguzman/LoRa_APRS_Tracker #236: T-Beam AXP2101-V1.2 not operating under battery power.

agfree, 2025-02-18:
After flashing the unit successfully with [LoRa_APRS_iGate](https://github.com/richonguzman/LoRa_APRS_iGate) firmware (and meshtastic) and the battery functioning properly, this firmware does not run under battery power at all.

S57PNX, 2025-02-19:
Generally, it does run on battery power, I have over 25 devices that do so. There must be something specific to your setup, for which we need to know much more about it - what board (picture is preferrable), which firmware did you install, how did you test it. More specifics please.

richonguzman, 2025-02-19:
Axp2101 power management chip of the Tbeam v1.2 usually when flashing needs a full reset 

So please take out usb and battery from the board.

Then plug in the battery again and it should work

agfree, 2025-02-19:
Thanks for the confirmation. I bought a three-pack of these on Aliexpress, and two operate properly. I wanted to explore any software options I might have missed, but it looks like this is just a problem with the hardware. I will begin the RMA process with Aliexpress.

closed 2025-02-19 by agfree