richonguzman/LoRa_APRS_Tracker #237: Buzzer sound question

oe9fwv, 2025-02-22:
I added Y44 buzzers to 2 trackers. One buzzer sounds somehow melodic, the startup sound is a three tone sound, the other one sounds much louder and is a single tone beep. The  trackers are both lilygo ttgo ESP32 boards, The first one is a version 1.0 ttgo, the second is programmed with v. 1.2. 
Which type of sound is the "correct" sound? 

73! Werner oe9fwv

richonguzman, 2025-02-22:
Werner, the 3 sounds is the way to go.

as is powered by only 3.3V it should not get so high sound as if it was powered by 5V

can you double check that the wirings on both modules are to the same pins, also that the pins definition on webUI - configuration page are the same and also that the modules are the same same same modules?

oe9fwv, 2025-02-22:
I connected IO25 to VCC and IO33 to I2C. The buzzers are the same batch of YL44. The only difference is the board type, V1.1 vs V1.2. 
All parameters on the display show correct values from the BME 280 and voltage from battery and USB charging.

richonguzman, 2025-02-22:
its really strange as the boards. (v1 or v1.1 or v1.2) should use the same pins and voltage connections for this and should behave the same (as I've tested here locally)