richonguzman/LoRa_APRS_Tracker #238: T-Deck some things to improve

SP9TKN, 2025-02-24:
Hello and welcome. 

RICARDO;
Thank you very much for the amazing T deck software, it works almost perfectly.
The functionality of the large screen and keyboard in the case of aprs and winlink is a new thing and a game changer at the same time. I support the development of this direction.
Below I am sending you some comments after a week of working with the device using this software. Please consider the comments positively and make corrections. I'm proud of your disposal, Adam Andrecki/SP9TKN

1. No change in backlight despite changing the min/max parameter
2. Confirming changes in menu/config by moving the trackball to the right results in no changes and a restart, but confirming by pressing causes changes to be saved without restarting
3. In the message option, when writing, only the first dozen or so characters (17) are visible, the line does not move/jump to a lower line.
4. in Winlink it is impossible to read the entire message because the text does not move/jump to a lower line
5. Winlink mail only displays the last 3 messages
6. In the message option, an error in displaying the keyboard before pressing the first character
7. No sound despite the speaker
8. The function of sending email with GPS coordinates does not work/I did not receive the message

9M2PJU, 2025-02-24:
Can you retry send email (GPS) again?

richonguzman, 2025-02-27:
> 1. No change in backlight despite changing the min/max parameter

on "Main" branch now is available to change between Low-Mid-Max brightness for Oled and Tft screens.

soon I will continue with the other fixes

SP9TKN, 2025-02-27:
Tnx, I will be waiting, I am still testing this amazing device under your direction

SP9TKN, 2025-03-01:
> Czy możesz spróbować ponownie wysłać wiadomość e-mail (GPS)?

Yes, problem is solved.

SP9TKN, 2025-03-07:
> > 1. No change in backlight despite changing the min/max parameter
> 
> on "Main" branch now is available to change between Low-Mid-Max brightness for Oled and Tft screens.
> 
> soon I will continue with the other fixes

Thanks, this feature working as well now.

Please, repair messages- 3 and 4 point.

Best regards, Adam

richonguzman, 2025-03-07:
just added better control of TFT screens too

SP9TKN, 2025-03-07:
yes, I tested it right now and it works great.

my favorite setting will be low

when writing aprs messages the lines of text do not scroll down, they do not scroll to the left.

when receiving messages the lines scroll down but one character remains covered on the right side.

i tried to fix the code myself but i am weak at it... i am a typical simple user.

richonguzman, 2025-03-07:
> yes, I tested it right now and it works great.
>
> my favorite setting will be low
>
> when writing aprs messages the lines of text do not scroll down, they do not scroll to the left.
>
> when receiving [...]

will try to fix this this weekend ;)

SP9TKN, 2025-03-12:
MOVED
Po zapytaniu o pogodę w sekcji raportu, tracker otrzymuje wiadomość od ca2rxu - 15, jak widać na załączonym obrazku, ale nie dekoduje odpowiedzi jako wiadomości.

Inne raporty sekcje pracy (szpital, policja, straż pożarna)

Pozdrawiam
TWÓJ TESTER
SP9TKN

[image]

SP9TKN, 2025-03-24:
can I ask you to send the ready compilation to sp9tkn@gmail.com?
please, it will make things much easier for me

SP9TKN, 2025-03-24:
ok, tomorrow I'll try to upload it to the tracker with the help of a friend and I'll let you know after the tests. Thanks!

SP9TKN, 2025-03-26:
sorry, I couldn't handle it for two evenings. I'll sit down with a programmer friend on Friday.

I'm better with a hammer than a keyboard...

I got this error.

In file included from src/LoRa_APRS_Tracker.cpp:22:
C:/Users/aandr/.platformio/packages/framework-arduinoespressif32@3.20009.0/libraries/BluetoothSerial/src/BluetoothSerial.h:18:10: fatal error: sdkconfig.h: No such file or directory

SP9TKN, 2025-03-29:
thank you for the changes you made. I'll still learn how to send software to the device correctly....

I edited the list with corrections, there was
winlink,
 keyboard display error,
 and wx report. unfortunately it still doesn't work...

but messages work great! thank you!

richonguzman, 2025-03-29:
> thank you for the changes you made. I'll still learn how to send software to the device correctly....
> 
> I edited the list with corrections, there was winlink, keyboard display error, and wx report. unfortunately it still doesn't work...
> 
> but messages work great! thank you!

if keyboard does not work, reset the T-deck and it should work and most of the problems will be solved

please test

SP9TKN, 2025-03-29:
[image]

richonguzman, 2025-03-29:
reset it until. its found! (sometimes need 2 rest from button) it should found it eventually

SP9TKN, 2025-03-29:
after pressing any button, the keyboard comes alive.

is there a possibility of improvement related to winlink and possible buzzer notifications?

SP9TKN, 2025-03-31:
yesterday, after putting the auxiliary digi for testing I discovered that the keyboard does not have the ' = ' sign.
it is not possible to execute some remote commands via the t deck plus tracker to the esp32 lora. is it possible to add ' = ' to the t deck keyboard?

SP9TKN, 2025-03-31:
after update messages a long message is displayed like this.

it's almost ok, but part of the message is still unreadable😃

[image]

richonguzman, 2025-11-05:
I'm finishing the full NRF52840 web Configurator and then i'll return to updates for Tracker and iGates ;)