richonguzman/LoRa_APRS_Tracker #239: Heltec V3 + GPS (DIY) minimum beacon time

PU5MFH, 2025-03-01:
Hi,

Set beacon time to mininum (5min), but its beaconing on default time (15min)
Tried with and without smart beaconing
actually, I need a lower time for coverage testing

richonguzman, 2025-03-02:
You would need to turn smartbeacon off and then set the beacon to less time than 15 min

But you can also send the beacon manually with user button when needed.

PU5MFH, 2025-03-03:
Thanks
smartbeacon set on fast moving (car/motorcycle), would be what beacon interval time? (sorry, I cant read the code to find myself)
didnt tested the newer version yet. last one I tested, when I was on the motorcycle, smartbeacon still send info like I was on foot, everytime.

richonguzman, 2025-03-03:
https://github.com/richonguzman/LoRa_APRS_Tracker/blob/main/src/smartbeacon_utils.cpp

car has: 
{120, 10, 60, 70, 100, 12, 10, 80}      // Car settings     = FAST

which means

at 10km/hr or slower --> 120sec between packets unless you change directions/angle
at 60km/hr or faste --> 60sec ....
in between it calculates the time accordingly to speed

closed 2025-03-24 by richonguzman