richonguzman/LoRa_APRS_Tracker #240: TDeck Plus No Display Backlight After 2.2.3 Update

barko33, 2025-03-07:
After updating my TDeck Plus to firmware 2.2.3 (6 March) there is no display backlight.  You can make out text and the firmware seems to be loaded and can be configured but the display is unreadable.

Went back to 2.2.2 (17 February) and everything was back to normal.

Reloaded 2.2.3 and the display backlight does not work.

Apprently an issue with 2.2.3 on the TDeck Plus.

violarulan, 2025-03-07:
Heltec Tracker v1.2 can reproduce this issue. 

v2.2.2 firmware works fine.

richonguzman, 2025-03-07:
Hi,

Did you change the brightness of the screen? Can you check this please?

richonguzman, 2025-03-07:
@violarulan @barko33 , V2.2.3 has been fixed .... I messed up with the brightness selection and now it should be fixed

please flash again again from webflasher V2.2.3 from 06.03.2025 to 07.03.2025

violarulan, 2025-03-07:
@richonguzman Just back home. Did a fresh re-install now and it works.🎉

barko33, 2025-03-07:
Just reflashed the TDeck Plus with the 07.03.2025 version and the display is now working correctly.

Thanks for the quick response!!

closed 2025-03-07 by richonguzman