richonguzman/LoRa_APRS_Tracker #244: Heltec wireless tracker: can't connect bluetooth

ruilvo, 2025-03-14:
Hi, 

Using this firmware on  a Heltec Wireless Tracker, I can't connect and use the Bluetooth TNC

What I have observed:

- On my Windows machines, it doesn't even pair.
- On my phone, it pairs.
- The BLE setting is not saved across boots. It starts always on. My Heltec doesn't have a battery. I always use external USB power.
- The problems persist even by turning off BLE (via the menu). 
- Using the phone, the Bluetooth icon does show up on the device, but the connection immediately drops whenever I try to do anything.

A screenshot from APRSDroid:

[image]

richonguzman, 2025-03-14:
ESP32s3 (as with Heltecs boards) only have BLE. At this moment in time APRSDroid only works with BT Classic and not with BLE. It gets added as a valid board for TNC but it will drop it as soon as it tries to work with it.

I'm currently working on a few ways to fix this...  but I have not a solution using the APRSDroid.

richonguzman, 2025-03-14:
which version of APRSDroid was the working with BLE?  any link?

as far as I know , BT/BLE only is phones and TNC is for usb-cable connected apps to a pc

ruilvo, 2025-03-14:
I think it's this fork: https://github.com/na7q/aprsdroid

richonguzman, 2025-03-14:
> So can this firmware TNC via USB/serial if I want to use my computer, then?

https://github.com/richonguzman/LoRa_APRS_iGate/wiki/107.-TNC-information

closed 2025-03-20 by ruilvo

ruilvo, 2025-03-20:
> which version of APRSDroid was the working with BLE? any link?
> 
> as far as I know , BT/BLE only is phones and TNC is for usb-cable connected apps to a pc

Releases are available under
https://na7q.com/aprsdroid-osm/