richonguzman/LoRa_APRS_Tracker #245: Lilygo TBeam v1.2 is does nothing after WEBUI  config/reboot

bertonumber1, 2025-03-15:
Hi , 
Thanks for bringing this development and your time spent on the project.

I have issues i will start with the main.

Main issue:

I flash the device with the latest firmware for the specified model Lilygo Tbeam v1.2. The device becomes unresponsive after WEBUI "Save"

Behaviour:

After flashing The behaviour is normal, AP is apparent and able to connect and WEBUI is accessed locally. 
OLED displays:
"NOCALL-10" 
"listening" 

Actual Behaviour (after reboot)
The device just gives up .
No AP /  Does not show up on network
The OLED just displays:
"LoRa APRS"
(Igate & Digi)
CA2RXU 2025.02.25

I have done a bit of reading and found a piece on editing Json files? I am guessing your webui omits the need for this . I am stumped. I am no stranger to flashing devices and embedded technologies so do not know what to do.

richonguzman, 2025-03-15:
Hi

After saving configuration the board reboots but without the wifiap (as it's not needed anymore)

The tracker should work without an issue this way.

If you need wifiap to change configuration, just press 3 times the user/middle button

bertonumber1, 2025-03-15:
Hi @richonguzman ,

I retraced my steps and taking your advice i i revisited the webui after starting fresh.
I  noticed i had placed decimal point in tx/rx frequency , when i removed those and rebooted . Everything now works perfectly .

Many Thanks for your time 

Rab MM7BVP

closed 2025-03-17 by richonguzman