richonguzman/LoRa_APRS_Tracker #246: Lilygo T-Deck Plus keeps rebooting when using APRSThursday

miodzicho, 2025-03-20:
I was able to Check In under APRSThursday menu, but any other option, like Join, KeepSubscribed leads to immediate reboot.

miodzicho, 2025-03-24:
I will ba happy to test.

richonguzman, 2025-03-24:
> I will be happy to test.

do you want to test VSCODE first before uploading to web flasher?

miodzicho, 2025-03-24:
Of course I can do it.

richonguzman, 2025-03-24:
please Test from Github (not yet on web flasher as it needs further testing)

miodzicho, 2025-03-25:
Not sure if it should send 4 times...

```
08:55:41 | DG6DXG-12>APLRT1,WIDE1-1::ANSRVR :CQ HOTG Happy #APRSThursday from LoRa Tracker 73! | -56 | 10
-- | -- | -- | --
08:55:46 | DG6DXG-12>APLRT1,WIDE1-1::ANSRVR :CQ HOTG Happy #APRSThursday from LoRa Tracker 73! | -55 | 10
08:55:51 | DG6DXG-12>APLRT1,WIDE1-1::ANSRVR :CQ HOTG Happy #APRSThursday from LoRa Tracker 73! | -57 | 13.75
08:56:02 | DG6DXG-12>APLRT1,WIDE1-1::ANSRVR :CQ HOTG Happy #APRSThursday from LoRa Tracker 73! | -58 | 14
08:56:06 | DG6DXG-12>APLRT1,WIDE1-1::ANSRVR :ackOI} | -57 | 12
``
```