richonguzman/LoRa_APRS_Tracker #247: Propuesta: Zonas de exclusión.

ea6ami, 2025-03-22:
Buenas tardes Ricardo,
Estaba pensando que podría ser útil para algunos usuarios implementar zonas de exclusión determinadas por coordenadas y radio, donde si el tracker detecta que está dentro de ella, no transmita su posición.

Lo veo útil por ejemplo, para instalaciones permanentes en el vehículo. 

Un saludo, 73.

richonguzman, 2025-03-22:
hola

me explicarías un poco mas la idea?

ea6ami, 2025-03-22:
Buenas, muchas gracias.

La idea sería, tal vez en el config, poder agregar un listado de coordenadas y radio de circunferencia (xx.xxxxxx, y.yyyyy, dddd) donde x=latitud, y=longitud, d=distancia (metros o km).

Si existen coordenadas configuradas, entonces el tracker, antes de transmitir la posición, calcula la distancia entre posición actual y las coordenadas configuradas. Si esa distancia es inferior a la establecida, entonces NO se transmite la posición.

Útil para establecer zonas donde no queremos ser trackeados (trabajo, casa...) sobretodo en instalaciones fijas como podría ser el coche, pudiendo hacer la instalación "desatendida", que se encienda al encender el coche y haga el track, despreocupandonos de acordarse de apagarlo antes de llegar al destino que no queremos que se comparta con el resto del mundo.

Un saludo, 73
Jaume

richonguzman, 2025-03-22:
Entiendo tu idea, pero igual me parece extraño que quieras compartir cosas y después no.

es como , APRS es un lugar sin encriptacion, publico donde todos ven todo, si no quieres  aparecer , lo apagas.

en código se puede hacer de todo, pero conceptualmente me parece que es muy especifico lo que mencionas.

de igual forma intentare ver que consigo

ea6ami, 2025-03-22:
Gracias por responder Ricardo,

No veo extraño querer compartir cierta información y a la vez asegurarse un mínimo de privacidad o establecer unos límites a la información compartida.

Por otro lado, no creo que sea una funcionalidad muy extraña donde trackers como TinyTrack,  Opentracker, tnc Mobilinkd la incorporan. Ya fuera del mundo de la radioafición, sistemas de trackeo público también disponen de lo mismo (strava, garmin,...).

Mi intención era aportar la idea por si podía ser interesante, si es inútil o no interesa entonces ni caso y se descarta.
Un saludo, 73

richonguzman, 2025-03-22:
seria tener una suerte de coordenadas donde no se trackea?

cuantos lugares deberían ser los "blockeados"? 1 , 2 , mas?

ea6ami, 2025-03-22:
Si eso es, puntos donde no se trackea, 
en mi caso particular, tengo interés en trackear, pero no me gusta compartir el punto exacto donde vivo o donde trabajo.

Cuantos ni idea, particularmente se que usaría mínimo 1 y máximo 2. Pero el interés particular no tiene porque ser el interés común, te diría de crear una lista como tienes para el wifi en el igate, pero no se si podría llegar a ser contraproducente en cuanto a rendimiento.

richonguzman, 2025-03-22:
por eso mismo preguntaba.... para entender toda la idea 

tendría que ser un cuadro con la posibilidad de hacer
Latitud + longitud + radio (en km) para que no sea tomado en cuenta?

ea6ami, 2025-03-22:
Si, eso es. 
Pensaba en un círculo, creo que es mas fácil su implementación. Con una circunferencia sería suficiente calcular la distancia entre las coordenadas actuales con las coordenadas del punto a omitir, si la distancia es menor al radio definido es que estamos dentro del área a omitir.