richonguzman/LoRa_APRS_Tracker #249: T-Deck cannot connect BLE to APRSDroid

miodzicho, 2025-03-25:
APRSDroid setup : 
TNC (Plaintext TNC2)
Bluetooth SPP
I have paired TDeck with my phone. BLE active on TDeck, but cannot connect from APRSDroid when pressing Tracking start.
Error is No connection to LoraTracker. Please turn on device or disconnect other bluetooth (nothing connected),

9M2PJU, 2025-03-25:
Did you use this version of aprsdroid https://na7q.com/aprsdroid-osm/  ?

miodzicho, 2025-03-25:
Correct, even reinstalled just today.
I know it's in German, but settings are self explanatory.
[image]

[image]

9M2PJU, 2025-03-25:
Try KISS mode

ruilvo, 2025-12-19:
Please see https://github.com/richonguzman/LoRa_APRS_Tracker/issues/260#issuecomment-3163115096 and https://github.com/richonguzman/LoRa_APRS_Tracker/issues/255#issuecomment-3163165783

TLDR: the NA7Q APRSDroid has a BAD implementation of BLE. It's not supposed to show the paired devices because pairing doesn't even make sense for BLE.