richonguzman/LoRa_APRS_Tracker #252: Power off / deep sleep mode for different trackers. GPS Eco mode

Jan-OK2, 2025-03-30:
1/Please can you clarify how it is with battery protection mode for different boards? Is it that only the T-Beam will turn off when the battery voltage drops below the set limit and turn back on when the USB port voltage comes back on (or pressed POWER ON button), while the others (Heltec modules, TTGO 2.1) go into deep sleep mode and the only way to wake them up is manually - press RST or disconnect/connect the battery?

2/How does GPS Eco mode work for each tracker mode configuration? I can understand turning off the GPS globally for use as a TNC. But when in Eco mode the GPS wakes up and goes to sleep while I am tracking? Sorry but I couldn't find it in the manual.
Thanks!

richonguzman, 2025-03-30:
hi Jan:

1) all  boards (with Power Management Chip or not) will go into deep sleep mode when going bellow "sleeping Voltage" if this is enabled. This makes the board sleep for a while , then wake up to see if voltage has gone up (like some re-charge) has been done, and if not, sleep again for a while.

Now , when a board is bellow certain voltage (and the board has Power Management Chip) it will start/restart right away. Boards without this (like Heltec v3) may need a reset button pressed as the voltage won't go up fast enough to reboot.

how are you powering it? (only usb or usb + battery?)

2) Sleeping GPS works by making the gps sleep for a variable time (smart beacon time dependent) and wakeup, check if it has moved from last registered place and if not, sleep again for a while. If it moved, send a beacon and sleep for a while based on the smartbeacon calculations on how far /fast he has moved.

Jan-OK2, 2025-04-01:
Hi, Ricardo,
thanks a lot for the answer. I'm trying different boards, so I can't answer in general yet. But only the T-Beam with PMU behaves exactly as expected - when the voltage drops below the set limit, and also with the power off button (even through config) it falls asleep and wakes up as soon as there is voltage on the USB - typically in the car (power from the ignition). 
I also have a Heltec wireless tracker board and a Heltec wireless stick (but some version 2.1, where only the FW for Heltec V2 works) and when I turn them off via config (user button) or when the voltage drops they cannot be woken up anymore. When USB is connected, only the LED lights up and the battery starts to charge, but the board sleeps on and there is no way to wake it up - only via RST. With the Wireless stick board I only measure as the external voltage connected to battery via divider and different pin- ADC input, it measures nonsense on the battery internal pin. But it works the same.
One more question - is there a difference in the kind of sleep mode when undervoltage and when "off" via config?

Thanks also for the explanation of GPS Eco mode. I tried it, but in the car it makes big jumps between positions (GPS sleeps too long) . Maybe at slow movement it can be useful.