richonguzman/LoRa_APRS_Tracker #253: HELTEC Wireless Tracker Low Battery Sleep

m1geo, 2025-04-04:
Hi.

I have my Heltec Wireless Tracker installed in my car with a LiPo battery. It works great powered from the boot 12V outlet to a USB-C cable.

When the external power is removed and the battery voltage drops below the set voltage (I'm using 3V) it sleeps. Great!

The issue is, it doesn't wake when external power is reapplied.

Is this a hardware limitation or a software bug?

Thanks, George. M1GEO.

[image]

richonguzman, 2025-04-04:
This board has no way to know when power is applied as it has not "power management chip" like some T-Beams have.

m1geo, 2025-04-04:
Understood. Is it possible to add a USB VBUS sense to GPIO (via a resistor PD [5V to 3.3V]) to detect the presence of USB VBUS and generate an interrupt to wake?

Alternatively, could those boards have a soft-wake function? As in, the CPU sleep for 1 minute interval, sample the battery voltage, and then either restart and run or return to sleep?

m1geo, 2025-04-04:
Another option would be to use an NPN transistor to trigger the nRST signal upon power connection. That sounds like it should work?

[image]

[image]

m1geo, 2025-04-04:
I'd prefer the software version, deep sleep for a few minutes at a time. It will be CPU dependant? But don't turn the screen or anything on, just wake, measure the voltage, and then make a decision. Everthing else at lowest power.

The hardware mod is a good option, but only as a last resort 🙂 but this feels very solveable in software with deep sleep and hardware timers. Perhaps there isn't a low power oscillator on the ESP32 (I don't know that architecture too well, more Arm stuff personally).

Thanks for your consideration and awesome project
George. M1GEO.

iw9grl, 2025-04-08:
George, did you do some trials with your suggestion to wake up the board? Are changes needed in the code?

Tnx
73 de IW9GRL

m1geo, 2025-04-09:
I did not, yet, try the mods. But it's a hacky solution. It'll 100% work, but it isn't elegant.

However, I _think_ this can be solved entirely in software? The Heltec Wireless Tracker is based around the ESP32-S3FN8 ([datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf)). The device supports 4 different sleep modes (active, modem-sleep, light-sleep, deep-sleep) with deep-sleep using 7uA.

The RTC stays powered up during deep sleep and can wake the CPU. There's also mention of the RTC ADC being powered all the time.

Conceptually, just set a timer to periodically wake things up, check the battery voltage, and then either: (a) sleep again if the voltage is too low, or (b) wake up and continue operation if the voltage is above the threshold.  Add some hysteresis between the sleep/wake voltages.

George, M1GEO.

Jan-OK2, 2025-04-10:
It could be very useful for some solar-based devices too!

richonguzman, 2025-04-10:
> It could be very useful for some solar-based digipeaters too!

i will share good news about this very soon (the same 7-10mA for solar digs in esp32 as NRF52840 boards ) ;)

m1geo, 2025-04-22:
Hey @richonguzman - don't forget me in the patent application :) lol

richonguzman, 2025-04-22:
> Hey [@richonguzman](https://github.com/richonguzman) - don't forget me in the patent application :) lol

wait a bit, i'm learning ultra low consumption process with BLE and NRF52840

maybe a NRF52840 BLE Keyboard for the trackers :) and much much more

m1geo, 2025-04-22:
That sound's cool! Just teasing you! I'm glad you've made some progress and to have helped with the gem of an idea :) hehe