richonguzman/LoRa_APRS_Tracker #254: T-Beam 1.1 915 sx1262 isn't working with firmware 2.2.4

JudiciousUneasiness, 2025-04-05:
I've been enjoying your firmwares on some 433 Tbeam 1.2s for a few months and decided to try it on my 915 TBeam 1.1s.  I have an 915 1.1 igate running your 2.3 igate firmware for a few days, but I cannot get the 2.2.4 tracker to work on my 915 tbeam 1.1 sx1262.  The error I'm seeing in the serial monitor is "Starting LoRa failed! State: -2" and the tracker screen doesn't progress beyond the "LoRa APRS" splash screen with your call.

[INFO][AXP192] init done!
[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[INFO][Main] Version: 2025.03.28
[ERROR][LoRa] Starting LoRa failed! State: -2

I tried to build and upload from Visual Studio Code but had the same result.  This TBeam has run meshtastic before this with no issues.

JudiciousUneasiness, 2025-04-05:
I just ran the webflasher with "T-Beam V1.0 (915MHz)" "V2.2.4 (28 March 2025)" and "First flash for Factory reset" selected and this is the full log after Install & web config:

[11:10:49]ets Jul 29 2019 12:21:46
[11:10:49]
[11:10:49]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:10:49]configsip: 0, SPIWP:0xee
[11:10:49]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:10:49]mode:DIO, clock div:2
[11:10:49]load:0x3fff0030,len:1184
[11:10:49]load:0x40078000,len:13232
[11:10:49]load:0x40080400,len:3028
[11:10:49]entry 0x400805e4
[11:10:51][INFO][AXP192] init done!
[11:10:55][INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[11:10:55][INFO][Main] Version: 2025.03.28
[11:10:55][WARN][Main] WebConfiguration Started!
[11:13:11]Got new config from www
[11:13:11]Saving config..
[11:13:11]Config saved
[11:13:12]ets Jul 29 2019 12:21:46
[11:13:12]
[11:13:12]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:13:12]configsip: 0, SPIWP:0xee
[11:13:12]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:13:12]mode:DIO, clock div:2
[11:13:12]load:0x3fff0030,len:1184
[11:13:12]load:0x40078000,len:13232
[11:13:12]load:0x40080400,len:3028
[11:13:12]entry 0x400805e4
[11:13:13][INFO][AXP192] init done!
[11:13:18][INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[11:13:18][INFO][Main] Version: 2025.03.28
[11:13:18][ERROR][LoRa] Starting LoRa failed! State: -2

JudiciousUneasiness, 2025-04-05:
Both the igate and this tracker have been configured for frequency 906700000, otherwise default LoRa BW/SF/CR/power settings

richonguzman, 2025-04-06:
Which board are you testing

Tbeam with sx1262 is only the v1.2

Saludos,

Ricardo / Richon
(Enviado desde mi celular a tubos)

JudiciousUneasiness, 2025-04-06:
I thought this is a 1.1 with the AXP192.  I will try the 1.2 firmware and see.
[image]

JudiciousUneasiness, 2025-04-06:
with the firmware from webflasher for tbeam 1.2 915 the result is the same:

```
[18:14:53]ets Jul 29 2019 12:21:46
[18:14:53]
[18:14:53]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:14:53]configsip: 0, SPIWP:0xee
[18:14:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:14:53]mode:DIO, clock div:2
[18:14:53]load:0x3fff0030,len:1184
[18:14:53]load:0x40078000,len:13232
[18:14:53]load:0x40080400,len:3028
[18:14:53]entry 0x400805e4
[18:14:55][ERROR][AXP2101] init failed!
[18:14:59][INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[18:14:59][INFO][Main] Version: 2025.03.28
[18:14:59][WARN][Main] WebConfiguration Started!
[18:17:42]Got new config from www
[18:17:42]Saving config..
[18:17:42]Config saved
[18:17:43]ets Jul 29 2019 12:21:46
[18:17:43]
[18:17:43]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:17:43]configsip: 0, SPIWP:0xee
[18:17:43]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:17:43]mode:DIO, clock div:2
[18:17:43]load:0x3fff0030,len:1184
[18:17:43]load:0x40078000,len:13232
[18:17:43]load:0x40080400,len:3028
[18:17:43]entry 0x400805e4
[18:17:44][ERROR][AXP2101] init failed!
[18:17:49][INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[18:17:49][INFO][Main] Version: 2025.03.28
[18:17:49][ERROR][LoRa] Starting LoRa failed! State: -2
```

richonguzman, 2025-04-06:
Show a picture of the board

JudiciousUneasiness, 2025-04-06:
this reply has a picture, does it show up for you? https://github.com/richonguzman/LoRa_APRS_Tracker/issues/254#issuecomment-2781177583

richonguzman, 2025-04-06:
This is a super rare board!

I never knew there was a T-beam v1 with SX1262 and NeoM8 gps!

JudiciousUneasiness, 2025-04-06:
I have a few of them that I bought several years ago, maybe they didn't sell very many of them?  Or mostly US sales.  Can I answer any questions about the board if it is so rare?  I can help test as well if it would help you.  It would be great to get APRS running on 915MHz in my area.

richonguzman, 2026-10-01:
added to Github V4.0.1 for testing (first VSCODE and if you validate it it will be added to webflasher)

closed 2026-10-01 by richonguzman