richonguzman/LoRa_APRS_Tracker #255: Bluetooth on android 15 (Pixel) and Supreme V3

salvadoriume, 2025-04-16:
Hello,
I have device supreme V3 and issue with Android 15 - tested on google pixel 7 and 9.
I see tracker but during pairing I got information that "cannot pair, code is invalid" - there wasn't even a popup to enter it.
I CAN pair tracker with laptop and other my phones on android 14.

I have other device tbeam v1.2 with meshtastic and it can be paired with my android 15.

Have somebody hit the same problem? 
Is is something on android side or tbeam?
Is there any other scenario I can test?

best regards SQ9MJ

richonguzman, 2025-04-16:
hi,

Supreme is a esp32s3 board, which has BLE. Android uses Aprsdroid which as far as I know has not implemented some BLE improvements which where proposed already. I'm trying to have a simple app for BLE and Android, so wait a little for this.

salvadoriume, 2025-04-16:
Hi,
thank you very much for a reply.
So my experience is not on app level, it is before aprsdroid etc, on system BT pairing phase.
Android 14 ok and android 15 cannot be paired - so I thought that maybe there is some change in communication between android and your code? Because I think that at beginning a problem need to be narrowed who is impacted - maybe it is not android but Pixel problem, but I don't have any other phone to test.

salvadoriume, 2025-04-17:
My google pixel phones have BLE, they are top android phones. But I read and now doubt about implementation of this BLE. The best would be to find some other BLE device and try to pair. Maybe I will restore meshtastic on supreme and check, I will let know

CraigBell, 2025-05-23:
I'm having a similar issue; I, too, have the Lilygo T-Beam Supreme. I was trying to use it with APRSdroid, but of course, that requires BT Classic, which I can't enable on the BT settings because the Supreme is only BLE!

richonguzman, 2025-05-23:
> I'm having a similar issue; I, too, have the Lilygo T-Beam Supreme. I was trying to use it with APRSdroid, but of course, that requires BT Classic, which I can't enable on the BT settings because the Supreme is only BLE!
> 
> Android phone is Samsung S22+ which in itself supports BLE, but won't work with APRSdroid.

i'm currently working on a way to use BLE + Android with Trackers

CraigBell, 2025-05-23:
Awesome, thanks! I look forward to testing it when it's ready. Thanks for your amazing work.

73 Craig - G8PY

frank-ydf, 2025-05-26:
Same issue here with a Pixel phone (Android 15) and a heltek v3.2 flashed with tracker firmware.
I had no luck in BLE and no luck in standard Bluetooth mode.
It continues to say "the passkey is invalid"

frank-ydf, 2025-05-26:
Ok, @salvadoriume !
I don't know how to do that but if I can help in some way, I'm happy to do that

ruilvo, 2025-08-07:
What I also found in the meanwhile is that the implementation of BLE that aprsdroid @na7q version is also buggy:

It asks you if you want to use BLE but then it shows a list of paired devices. So you feel tempted to try to pair the device.

This is absolutely incorrect behaviour on the app side. If you're trying to use BLE you do not pair any devices so you should use the Android APIs that show you nearby devices, not paired devices.

So, my suggestion would be that  if possible, make the device stop showing as a Bluetooth classic device, ready to pair, when really on BLE mode, and for @na7q to also fix the fork of aprsdroid.

And, of course, if possible, make toggling between BLE and Bluetooth classic work.

"Bluetooth baixa energia" is BLE 
[image]

But then it shows paired devices, which is wrong. 
[image]

armegia, 2026-09-05:
I cannot even dream of touching Android so I am using NA7Q build untouched. However with Codex I could implement some kind of pairing in the firmware - and tested it on my recently acquired T-Beam Supreme V3. I also added a couple new features related to logging/debugging, clear the paired devices, add some small labels to the GUI configurations (possibly this should be removed but it seemed a convenient way to test whether codex made sense). My fork: https://github.com/armegia/LoRa_APRS_Tracker/tree/pr/diagnostics-ble-bonding

--
Antonio EA4HXZ

richonguzman, 2026-09-05:
Hola Antonio
Why is this needed? to make APRSDroid work ?

armegia, 2026-09-05:
> Hola Antonio, why is this needed? To make APRSDroid work?

It actually works without these changes, but Android asked me to authorize every connection. The firmware was not creating a persistent BLE bond. After these changes, I paired once using a random PIN shown on the tracker, and APRSDroid reconnects automatically even after a tracker reset.

The only APRSDroid version I use is NA7Q’s build; I could not make the official version work with BLE a few months ago and I have not tried again.

I mostly used this as an exercise in embedded and AI-assisted development. The changes are:

- Authenticated, persistent BLE bonding with a random on-screen PIN.
- A Web UI action to clear stored BLE bonds and a few other changes in the webui, all should be documented in the PR draft I am linking below if you want to review.
- Thread-safe, runtime-configurable serial logging.
- Additional radio diagnostics and small board-aware Web UI improvements.

I tested pairing, automatic reconnection, persistence across resets, and bidirectional APRS messaging on a T-Beam Supreme V3.

Full explanation, test results, limitations and attribution:  
https://github.com/armegia/LoRa_APRS_Tracker/blob/review/diagnostics-ble-bonding-notes/PULL_REQUEST_DRAFT.md

Thanks

richonguzman, 2026-09-05:
good!

I will read this!!

On iOS bonding "stays" but on Android I don't know, but I have a beta test new app for android to test this bonding issue that maybe you want to test?

armegia, 2026-09-05:
That would be great. However, bear in mind that I have no idea what I am doing and Codex wrote all the code; I tried my best to make sure it made sense but mostly I reported bugs and kept on testing while Codex did the work. I have an older android 11 and a pixel 8 (android 17).

armegia, 2026-09-05:
I am doing a code review now and will propose a cleaner set of changes later today. Codex did a half assed work in some places. I will have it tighten some bolts and come bacck