richonguzman/LoRa_APRS_Tracker #257: T-Beam 1.2: OLED 1.3" not working

andrea-dandrea, 2025-04-26:
Hello, I just unpacked my brand new T-Beam 1.2 and gave it a go with both LoRa APRS Tracker and Meshtastic.
While Meshtastic seems to work fine, there is an issue with the OLED screen in LoRa APRS Tracker.
The screen is garbled, only one line kind of works as you can see in the updated videos.
I tried several firmware versions, and did some configuration from the webserver, but could not find anything related to the display to improve its behavior (the time out works, for instance) and I've not been able to make it work.
Not sure if it depends on the version of my OLED screen; it is the 1.3" with 3 buttons and SH1106, which to my knowledge is supported.
Any help would be welcome!!

BR,
Andrea

https://github.com/user-attachments/assets/ab2ac94c-efe5-4372-bf79-8d2246a72f94
https://github.com/user-attachments/assets/4e5723b9-977b-4d4b-8c77-c5dfeca091a2

richonguzman, 2025-04-26:
Hello!

this screen works but needs a few tricks "for now"

can you use Platformio to update the firmware with the small mod (so it will work right away?)?

andrea-dandrea, 2025-04-26:
Hola Ricardo,
gracias por la contestación tan rapida!!
I'm not that good with the developement environment, but I can give it a try with some guidance.
I have installed in my PC the ESP-IDF environment to do some tests with M5Stack stuff, will that be useful for this?

richonguzman, 2025-04-26:
you would have to download the repository zip file. Download VSCODE and Platformio.

import the repository to VSCODE with Platformio, choose your board and change a comment in a single line of it.

i can guide you!

Si tienes Telegram buscase con el mismo usuario que uso aqui en Github y me escribes y te guío