richonguzman/LoRa_APRS_Tracker #260: Can't disable BLE

ruilvo, 2025-05-23:
Hi,

I have an Heltec Wireless Tracker.

After a firmware reset / full clean flash, BLE shows up as disabled in the Web GUI.
If I turn it on, it works.

But, any time after, if I try to **disable** BLE, it doesn't work.

I disable BLE, save. The device reboots, I connect to AP again and BLE shows up as enabled.

richonguzman, 2025-05-23:
will check this out very soon

sm3sgp, 2025-07-24:
I can confirm some of the settings done done locally in the tracker does stay after a reboot, and if you configure via the Web-gui you get a different behavior.
I have seen some mismatch when changing Bluetooth settings and the Display timeout. I will see if I can pinpoint my findings better.
I also have the Heltec Wireless Tracker.

richonguzman, 2025-07-24:
ok! will double check a few things and let me know how it goes on your side

ruilvo, 2025-08-06:
For me it does retain the settings even with a reboot. 

Now, I also changed phones and on my Pixel phone it refuses to connect with an error message saying "incorrect passkey or PIN". 

If I flash another firmware on it, e.g., meshstatic, it works fine. 

So really something  weird going on with the bluetooth implementation.

ruilvo, 2025-08-06:
The logs say 

```
[14:49:08][INFO][BLE] BLE Client Connected
[14:49:14][INFO][BLE] BLE client Disconnected, Started Advertising
[14:49:15][INFO][BLE] BLE Client Connected
[14:49:20][INFO][BLE] BLE client Disconnected, Started Advertising
```

The phone shows:

[image]

The device is not saved on the phone, I already reset bluetooth on the phone as well.

ruilvo, 2025-08-07:
One more day, one more information.

In hindsight, it's obvious: you can't pairaa BLE device, since that's a feature of traditional Bluetooth.

Of course here the problem is that the device shows up as available for pairing when it probably shouldn't. Maybe some mix between Bluetooth classic and BLE that is not supported either by the Heltec WT or by this firmware... I have many devices at home which have both BLE and BT classic and they show up as different devices on my phone.

So, what I understand so far: 

- the device it totally working as a BLE device.
- can't turn off BLE
- not being able to pair is a problem of the device being on BLE mode but still broadcasting BT classic.

What do I have to show for it?
I've tried [this app](https://app.lora-aprs.pl/download) and I could connect with the device totally fine on BLE mode. 

[image]

[image]