richonguzman/LoRa_APRS_Tracker #277: Problem establishing Bluetooth TNC connection (while also standalone tracker)

Uberseer, 2025-07-24:
I have LoRa_APRS_Tracker v2.2.4 installed on a heltec_wifi_lora_32_V3_GPS.  It works fine as standalone, but I can't get it to connect to my APRSdroid app.

My tracker_conf.json contains:

```
"bluetooth": {
		"active": true,
		"deviceName": "LoRaTracker",
		"useBLE": false,
		"useKISS": true
	},
```

and I have the device paired with my tablet.  In APRSdroid when I click start tracking, i see a message that it is trying to connect to SPP service, then fails with:
`Could not connect to LoRaTracker! Okease turn on the device and close other Bluetooth connections.`

I can confirm no other Bluetooth connections.  I have APRSdroid setup to use TNC(KISS), connection type "Bluetooth SPP", blank channel, LoRa Tracker device is selected.  Why can't I make a BT Classic connection to the LoraTracker as a KISS TNC?

richonguzman, 2025-07-24:
Hi,

the regular version of APRSDROID does not connect over BLE (which is the only type of Bluetooth the Heltec V3 board has) (so, this board has no BT-Classic possibility)

there are a few variations of apps that could work over BLE (like NA7Q version in beta testing)

Uberseer, 2025-07-24:
I didn't realize that!  Ok, I switch useBLE to true on the tracker, and useKISS true:
```
"bluetooth": {
		"active": true,
		"deviceName": "LoRaTracker_VE3OCG",
		"useBLE": true,
		"useKISS": true
	},
```
and re-uploaded, then install NA7Q's APRSdroid.  In APRSdroid I left protocol as KISS (to match my tracker settings), connection type to BLE, selected device.  Then start connection...but it still connect to LoRaTracker.

Is there anything else I need to change?

Uberseer, 2025-07-24:
Still no luck...let me back up a bit.  I did in fact rename my device as shown above...but on the tablet I still only see the old BT name.  And when I try to pair with it from Android settings I see:
`An app is needed to use this device`

Is that normal?  Do I have to pair the heltec to Android BEFORE entering APRSdroid?  The "TNC Bluetooth Device" list in APRSdroid does not offer me my tracker, and in Bluetooth Settings I see "LoRaTracker" but pairing fails with "couldn't connected"

Wondering if something is cached somewhere.  I have rebooted both devices.  Still stuck...