richonguzman/LoRa_APRS_Tracker #290: Heltec Wireless Tracker can't beacon; shuts off screen.

RhinoLance, 2025-09-25:
Board: Heltec Wireless Tracker (433 MHz)
Firmware: V2.3 (11 August '25)

Whenever the tracker tries to beacon its position, it shuts off the screen, requiring a press to re-activate it, at which point it tries to beacon again, and so it goes.  The beacon messages are never received by the local iGate.

I don't believe it's restarting, as when I re-activate the screen it still has the GPS position.  A restart would revert to a position of 0,0.

There is no difference if powered only from battery, or plugged into a USB power source.

Any thoughts?

Lance . .

I have provided a video demonstrating the behaviour.  This is from a cold start with a position of 0,0.  As soon as a position is established, it tries to beacon.

[image]

Szetya, 2025-09-29:
> Board: Heltec Wireless Tracker (433 MHz) Firmware: V2.3 (11 August '25)
> 
> Whenever the tracker tries to beacon its position, it shuts off the screen, requiring a press to re-activate it, at which point it tries to beacon again, and so it goes. The beacon messages are never received by the local iGate.
> 
> I don't believe it's restarting, as when I re-activate the screen it still has the GPS position. A restart would revert to a position of 0,0.
> 
> There is no difference if powered only from battery, or plugged into a USB power source.
> 
> It appears this is as a result of having Eco GPS enabled. I was under the impression that just the GPS should turn off, not the screen too.
> 
> Any thoughts?
> 
> Lance . .
> 
> I have provided a video demonstrating the behaviour. This is from a cold start with a position of 0,0. As soon as a position is established, it tries to beacon.
> 
> [image]

The fundamental problem is that, according to the concept, VEXT_CTRL (Voltage EXTernal ConTRoL) would turn off the GPS power supply. However, VEXT not only turns off the GPS, but also the display backlight and the LoRa radio power line. That is why all this is happening. 
I already pointed out to @richonguzman  during the modification that this was not good. But my message fell on deaf ears. Turn off Eco mode and it will be fine. 

Translated with DeepL.com (free version)

RhinoLance, 2025-09-29:
Thanks for the explanation @Szetya.  Now I know what is happening and why.

Szetya, 2025-09-29:
> Gondolom, ennek semmi köze a GPS eco módjához, hanem az a probléma, hogy a panelnek több szűrésre van szüksége (100uF) a 3,3 V-on.

[image]

richonguzman, 2025-09-29:
> > Gondolom, ennek semmi köze a GPS eco módjához, hanem az a probléma, hogy a panelnek több szűrésre van szüksége (100uF) a 3,3 V-on.
> 
> <img alt="Image" width="2000" height="731" src="https://private-user-images.githubusercontent.com/22783982/495232364-4e6ebf86-7080-4f68-b33d-ebce2c44b034.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxNTY5NzgsIm5iZiI6MTc1OTE1NjY3OCwicGF0aCI6Ii8yMjc4Mzk4Mi80OTUyMzIzNjQtNGU2ZWJmODYtNzA4MC00ZjY4LWIzM2QtZWJjZTJjNDRiMDM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDE0Mzc1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlZTFhNTkxMDM3Y2JhZDdiZWVhMTE2NmQzZjg2OGQ0MjdiMDA5Nzg0MDkxMjlhYTUwNTZkMGE3OWJkMGQ3MGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ooeR2K50a2Dhgi0-RGhmUrilafVENTtwKD43eU5-t74">

?????

richonguzman, 2025-09-29:
on the GPS Eco Mode  (idea) , which is not a finall idea, the goal was to put GPS off and use less current... and it works for this but turns of the screen (known behavior)

Szetya, 2025-09-29:
> on the GPS Eco Mode (idea) , which is not a finall idea, the goal was to put GPS off and use less current... and it works for this but turns of the screen (known behavior)

The screen VDD is also controlled by Vext. If you turn off Vext, the display power supply will be cut off.  As well as the GPS power supply.
This includes all communication with the GPS and the display.
[image]

richonguzman, 2025-09-29:
exactly, this is the known behavior for this board

RhinoLance, 2025-09-29:
OK, after some more testing I have results consistent with what's been explained here.
There must have been an issue with the iGate when initially testing, as my beaconing is now received with Eco GPS on and off.  So it always transmits OK.

What I expect is happening in the video then is expected.  After it beacons, the screen turns off.  When I press the button to "wake" it, it's the beaconing button, so it beacons again, and turns the screen off.  And so it repeated.

I have since disabled the Eco GPS, and I still get more than enough life out of a battery so am happy.

If the screen turning off is expected behaviour, it may be worth mentioning it in the documentation to save others any confusion.

Thanks for all the responses and explanations.

richonguzman, 2025-09-29:
> OK, after some more testing I have results consistent with what's been explained here. There must have been an issue with the iGate when initially testing, as my beaconing is now received with Eco [...]

will add it the soon-coming Tracker Manual

the tracker in gps-eco-mode should work with 2 clicks to enter menu as regularly and only when out of the menu's it will return into Display+GPS eco mode

test it if you like ;)